winter-telescope/mirar #41: Processing candidates to Fritz

sukishore12, 2022-07-25:
If Fritz is down, we run into the issue that candidate cannot be processed/annotations cannot be updated. To ensure this doesn't cause the pipeline to break:

- Once candidates table is in the database, have a field `is_fritz_processed`. Query everything that hasn’t been submitted to fritz and feed them to SendToFritz processor
- If Fritz is down, update the `is_fritz_prcessed` field to False for that candidate, wait 30 seconds (Fritz goes now in 30 sec stretches), then continue
- Successfull Fritz processing/updating, sets `is_fritz_prcessed` field to True
- Query `is_fritz_processed` again for False entries after going through all cands

robertdstein, 2022-07-25:
I am very worried about this step in general. I think Fritz could easily be down for long stretches, so we should think carefully about the right behaviour in those cases. How often should we retry? Refreshing every 30 seconds is likely to lead instability, because a backlog will build up. (Remember there are many other places/codes/people that use Fritz, and they also refresh/retry repeatedly if Fritz is down).

robertdstein, 2022-07-25:
(To be clear, everything you wrote sounds reasonable. It's just more a general worry I have)

virajkaram, 2022-07-25:
Long-term outages on Fritz will have to be dealt separately (I can see scenarios where we stop sending candidates to fritz altogether until the team figures out what is wrong). Fritz also has several short-term outages per day (502 errors that go away on their own in some time, ~30 sec) that we will most certainly encounter during nightly runs of the pipeline. The purpose of this issue is to make sure that these Fritz errors don't crash the pipeline, and also that we don't miss out on sending candidates because of these.  This architecture does not exist right now.

I don't think we need to resubmit fritz queries every 30 seconds. We just need a queue-like infrastructure from which sources get submitted to Fritz. If a source is successfully submitted, it gets removed from the queue, otherwise it stays in it. Every time the SendtoFritz processor is run, it will try to send everything in the queue (or the top xx sources from the queue, depending on how much load fritz can handle). Currently, this process will happen every time a new batch of images is processed, so short-lived fritz errors will negated when Fritz comes back online. 

As we don't have much control over Fritz errors, maybe it makes sense to move the SendtoFritz processor out of the pipeline and into a watchdog? Then we would be able to set the time interval between successive fritz queries as well.

virajkaram, 2022-11-04:
We copied the kowalski implementation for doing this.

closed 2022-11-04 by virajkaram