winter-telescope/mirar #111: Weird Problem with the Sextractor Module

robertdstein, 2022-08-30:
When trying to reduce data, I am running into a weird error:

```
Error for processor winterdrp.processors.astromatic.sextractor.sextractor at 2022-08-30 12:10:28.784147 (local time): 
   File "/Users/robertstein/Code/winterdrp/winterdrp/processors/base_processor.py", line 127, in base_apply
    batch = self.apply(batch)
  File "/Users/robertstein/Code/winterdrp/winterdrp/processors/base_processor.py", line 199, in apply
    images, headers = self._apply_to_images(images, headers)
  File "/Users/robertstein/Code/winterdrp/winterdrp/processors/astromatic/sextractor/sextractor.py", line 179, in _apply_to_images
    header[sextractor_checkimg_keys[checkimg_type]] = checkimage_name[ind]
KeyError: 'NONE' 
  This error affected the following files: ['SUMMER_20220824_204552_Camera0.resamp.fits'] 
This error was not a known error raised by winterdrp. 
```

Beyond the typos, why would the code be trying to get an entry marked "None"?

virajkaram, 2022-08-30:
Is this for only one image or for all images?

robertdstein, 2022-08-30:
Never mind, I solved this. The summer pipeline was using 'None' not None here (https://github.com/winter-telescope/winterdrp/blob/180177bd7927dae57b0ce1ef72a9dcec8916490e/winterdrp/pipelines/summer/summer_pipeline.py#L432), which I updated. A copy/paste from an older version of the code, maybe?

commit 9d485e91 (2022-08-30): Recycle standard reduction for realtime, fix #111
commit 126e9a1e (2024-04-17): Recycle standard reduction for realtime, fix #111
closed 2022-09-01 by robertdstein