winter-telescope/mirar #301: [BUG] poetry install winterdrp fails due to pyfftw install issue

broulston, 2023-02-27:
**Describe the bug**
Installing winterdrp fails on `poetry install`. The issue seems to be with installing pyfftw. 

However, using `conda install pyfftw` works and installs with no issues. 

```
 ChefBuildError

  Backend subprocess exited when trying to invoke build_wheel

  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-11.0-arm64-3.11
  creating build/lib.macosx-11.0-arm64-3.11/pyfftw
  copying pyfftw/config.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw
  copying pyfftw/_version.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw
  copying pyfftw/__init__.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw
  creating build/lib.macosx-11.0-arm64-3.11/pyfftw/builders
  copying pyfftw/builders/builders.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/builders
  copying pyfftw/builders/__init__.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/builders
  copying pyfftw/builders/_utils.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/builders
  creating build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/cache.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/__init__.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/scipy_fft.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/dask_fft.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/numpy_fft.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/scipy_fftpack.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  copying pyfftw/interfaces/_utils.py -> build/lib.macosx-11.0-arm64-3.11/pyfftw/interfaces
  UPDATING build/lib.macosx-11.0-arm64-3.11/pyfftw/_version.py
  set build/lib.macosx-11.0-arm64-3.11/pyfftw/_version.py to '0.13.1'
  running build_ext
  DEBUG:__main__:Link FFTW dynamically
  DEBUG:__main__:Compiler include_dirs: ['/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/include', '/Users/benjaminroulston/Dropbox/Research/Coding/PYTHON3_ENVS/winter_env/include/python3.11']
  DEBUG:__main__:3.11.0 | packaged by conda-forge | (main, Jan 14 2023, 12:26:40) [Clang 14.0.6 ]
  DEBUG:__main__:Sniffer include_dirs: ['/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/include', '/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/pyfftw', '/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/lib/python3.11/site-packages/numpy/core/include', '/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/include']
  DEBUG:__main__:objects: []
  DEBUG:__main__:libraries: []
  DEBUG:__main__:include dirs: ['/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/include', '/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/pyfftw', '/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/lib/python3.11/site-packages/numpy/core/include', '/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/include']
  /var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/pyfftw-s8c69wv8/None.c:1:10: fatal error: 'fftw3.h' file not found
  #include <fftw3.h>
           ^~~~~~~~~
  1 error generated.
  WARNING:__main__:Compilation error: command '/usr/bin/clang' failed with exit code 1
  DEBUG:__main__:clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/benjaminroulston/Dropbox/Research/Coding/PYTHON3_ENVS/winter_env/include -arch arm64 -fPIC -O2 -isystem /Users/benjaminroulston/Dropbox/Research/Coding/PYTHON3_ENVS/winter_env/include -arch arm64 -I/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/include -I/private/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpsmidimrs/pyFFTW-0.13.1/pyfftw -I/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/lib/python3.11/site-packages/numpy/core/include -I/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/include -I/var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/tmpouvgg7av/.venv/include -I/Users/benjaminroulston/Dropbox/Research/Coding/PYTHON3_ENVS/winter_env/include/python3.11 -c /var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/pyfftw-s8c69wv8/None.c -o /var/folders/4n/lyhfznl925q5wvhw0gjjdrd00000gn/T/pyfftw-s8c69wv8/None.o

  DEBUG:__main__:Checking with includes ['fftw3.h']...no
  error: Could not find the FFTW header 'fftw3.h'

  at ~/Dropbox/Research/Coding/PYTHON3_ENVS/winter_env/lib/python3.11/site-packages/poetry/installation/chef.py:152 in _prepare
      148│
      149│                 error = ChefBuildError("\n\n".join(message_parts))
      150│
      151│             if error is not None:
    → 152│                 raise error from None
      153│
      154│             return path
      155│
      156│     def _prepare_sdist(self, archive: Path, destination: Path | None = None) -> Path:

Note: This error originates from the build backend, and is likely not a problem with poetry but with pyfftw (0.13.1) not supporting PEP 517 builds. You can verify this by running 'pip wheel --use-pep517 "pyfftw (==0.13.1) ; python_version >= "3.8""'.
```

**To Reproduce**
Steps to reproduce the behaviour:
```zsh
conda create -n winter_env python=3.11
conda activate winter_env
pip install --upgrade pip
pip install poetry
poetry install

```

**Expected behaviour**
`poetry` should install `winterdrp` without error.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.2

**Additional context**

broulston, 2023-02-28:
This seems to be a mac arm vs. x86 issue. Running a terminal in x86 through Rosetta all the install commands work without issue.

robertdstein, 2023-04-22:
Yes, I faced this too. I think the basic issue is that the current conda build only exist for `osx-64` but not `osx-arm64`. We can try making an issue/requesting extra builds, but in the meantime the interim solution is to create a conda environment that uses the `osx-64` builds: https://stackoverflow.com/questions/71515117/how-to-set-up-a-conda-osx-64-environment-on-arm-mac

robertdstein, 2023-04-27:
For future reference, the proposed interim solution for Mac users is:

```
conda create -n winter_env
conda activate winter_env
conda config --env --set subdir osx-64
conda install python=3.11
conda activate winter_env
pip install --upgrade pip
```

We can revisit when there are osx-arm64 wheels (i.e resolution of https://github.com/pyFFTW/pyFFTW/issues/352), which will probably not happen before Github release M1 runners (https://github.com/github/roadmap/issues/528).

closed 2023-04-27 by robertdstein