winter-telescope/mirar #333: [BUG] Ensure Swarp does not propagate astrometric headers

virajkaram, 2023-04-08:
**Describe the bug**
Currently, the Swarp processor propagates header keywords from parent images into final images. Sometimes, this results in propagation of stray astrometric keywords that swarp does not override.

This causes problems sometimes, and is undesirable behaviour.

virajkaram, 2023-04-09:
Partly (or fully?) addressed by PR #336.
@saarahha when you get the chance, can you please confirm that [this](https://github.com/winter-telescope/winterdrp/blob/1c7484ca9912ab003d211503378e924c6f63446c/winterdrp/paths.py#L209) includes all astrometry keywords that astrometry.net can add to the header?

saarahha, 2023-04-09:
@virajkaram There are some other astrometry.net keywords that are still missing from the file you linked.
In an a-net output file header, underneath `COMMENT --Start of Astrometry.net WCS solution--` there are still:

WCSAXES
EQUINOX  
LONPOLE  
LATPOLE  
CUNIT1  
CUNIT2  
IMAGEW  
IMAGEH   
AP_0_0   
AP_0_1   
AP_0_2   
AP_1_0   
AP_1_1  
AP_2_0  
BP_0_0  
BP_0_1   
BP_0_2   
BP_1_0   
BP_1_1   
BP_2_0

virajkaram, 2023-04-14:
Fixed by #341

closed 2023-04-14 by virajkaram