winter-telescope/mirar #362: [BUG] printed None and None directory made for output

broulston, 2023-05-03:
**Describe the bug**
When running a pipeline (e.g. `python -m winterdrp -p sedmv2 -c default_stellar -n 20221223`) a set of `None` are printed in the output at the very beginning. Not sure if these are related to the MAX_N_CPUs or not. 

```
(winter_env) ➜  winterdrp git:(main) ✗ python -m winterdrp -p sedmv2 -c default_stellar -n 20221223
None
None
None
None
None
None
None
None
Using cache /Users/benjaminroulston/Dropbox/Research/Projects/SEDMv2/OUTPUT_DATA/winterdrp_temp/tmpmag17gee
winterdrp.pipelines [l 22] - INFO - Found sedmv2 pipeline
```

Potentially related is the fact that sometimes temporary files are saved to a `None` folder in the `OUTPUT_DATA_DIR` instead of the expected cache directory. 

**To Reproduce**

Run a pipeline `python -m winterdrp -p sedmv2 -c default_stellar -n 20221223`

virajkaram, 2023-05-03:
The reason for the `None` printing is some stray print statements that somehow made their way to main. I've fixed this in a branch I am working on, will merge it soon. 

Re the `None` folder, that happens when a processor is initialized without setting the night directory. Can you give some more info on what kind of temporary files are being saved in it?

broulston, 2023-05-03:
Oh ok, I had searched for some print statements but hadn't found any that looked like they might be the cause. Will wait for your update!

virajkaram, 2023-05-09:
First part (at least) solved by #369

broulston, 2023-05-12:
> @broulston can you check whether this is solved?

Yes, confirm no more `None` printed when run, and currently no more `None` output directory made. Not sure if you intended to fix that, or if it is really fixed. But currently running it through as I did before it does not make that extre pipeline directory. 👍

virajkaram, 2023-05-15:
Ok great. Closed by #369

closed 2023-05-15 by virajkaram