winter-telescope/mirar #371: poor sedmv2 photometry

saarahha, 2023-05-11:
**Describe the bug**
There is some room for improvement on sedmv2 photometry outputs - some bogus/saturated/edge sources are messing with the zero-point calculation in `PhotCal.py`. To address this, when calculating the zero-point we should:

- Exclude PS1 reference sources that are flagged as saturated (this should be the default behavior for all instruments)
- Exclude sources detected on the edge of the IFU mask for sedmv2

broulston, 2023-05-12:
@saarahha for the second point about excluding sources at the IFU mask edges, are you planning on implementing that using sourceextractor? I know the PSF photometry is failing primarily because it's trying to build PSF models which include bogus sources at the IFU mask edge. The sourceextrator catalog is remade for the PSF step so any fix would likely need to be in the sourceextractor processor.  I was going to look at this but if you already are great! Let me know if you need any help! 😀

saarahha, 2023-05-12:
@broulston Great minds think alike... I will give this a try today within sourceextractor, and I'll let you know!

saarahha, 2023-06-23:
* should also address the recent issue where the first frame of SEDMv2 MEFs looks faulty. throw the first frames away after MultiExtParser?

virajkaram, 2023-06-30:
Reopening because this was reverted by #412 . Will be fixed in a separate PR.

robertdstein, 2023-06-30:
Sorry Saarah!

virajkaram, 2023-06-30:
It was @robertdstein 's idea

PR #426 (2023-06-26): Better photometry for SEDMv2 (and beyond...)
This PR mainly changes the SEDMv2 pipeline, by:
- ensuring the IFU mask (made with `MaskPixels`) is carried down the pipeline, and incorporated into the later masks (like the one made by `Swarp`). This should reduce the chance that a half-covered-by-IFU source appears in `SExtractor` catalogs.
- adding the option to ignore the first frame in a MEF, since recent SEDMv2 data has faulty first frames

This PR also adds a new default behavior:
- when using PS1 as a photometric reference catalog, exclude PS1 sources flagged as "SATURATED". This change is  [... cut]