winter-telescope/mirar #434: Installation docs update.

jamiesoon, 2023-07-03:
When following the installation docs for installation an issue may arise when using poetry and access to a keyring.

This is a known poetry issue:
https://github.com/python-poetry/poetry/issues/1917

and can be solved by using:
"export PYTHON_KEYRING_BACKEND=keyring.backends.null.Keyring"

None of the required packages should require the keyring so this should be added to the installation docs.

jamiesoon, 2023-07-03:
Also missing the example command:
"import mirar"
when checking if the installation is successful.

robertdstein, 2023-07-04:
Thanks for sharing this @jamiesoon. Can you post the exact error message/traceback that you get? We can either post that in the docs, or just suggest that people search our github issues page to find any known/solved ones. And in future would you be able to open separate Github issues for separate problems? I can fix the second problem you noticed immediately in a PR.

jamiesoon, 2023-07-12:
Unfortunately not as I didn't save the error message at the time.
I've tried to recreate it but I currently don't have access to a clean machine with the right conditions.
I'm fairly certain the error would arise in a new installation of Debian/Ubuntu that runs a desktop environment.

robertdstein, 2023-07-12:
I'm not sure if that's true. At least, for the CI we pull up a clean ubuntu-2204:

https://github.com/winter-telescope/mirar/blob/356bf25fb9573f90a66bb4989e5e83c4225268ee/.github/workflows/continous_integration.yml#L24

and then install pip/poetry from scratch, without the PYTHON_KEYRING_BACKEND exporting. So perhaps it depends specifically on your platform version?

In any case, I'm not sure we can do much more here. I will close for now, but if you (or anyone else) encounters it again please reopen and we can document for it!

closed 2023-07-12 by robertdstein

jamiesoon, 2023-08-11:
I've encountered this issue again when updating via poetry with the following error

```
Installing dependencies from lock file

Package operations: 11 installs, 75 updates, 0 removals

  • Installing rpds-py (0.9.2): Failed

  KeyringLocked

  Failed to unlock the collection!
```

Had to do "export PYTHON_KEYRING_BACKEND=keyring.backends.null.Keyring" again.