winter-telescope/mirar #469: [BUG] check_file_is_complete always fails for MEF images

virajkaram, 2023-07-10:
**Describe the bug**
When using the current check_file_is_complete function for all MEF images, it always fails. I don't think I fully understand the logic of this function.

**Expected behaviour**
It will not fail and say that the MEF image is complete

virajkaram, 2023-07-10:
Fixing by setting it to astropy logic (sets file incomplete if file.tell() > file.size())

virajkaram, 2023-07-10:
https://github.com/astropy/astropy/blob/1ef68aa829db94b1c506afbb7f6fc2faeaf4eee5/astropy/io/fits/file.py#L430

robertdstein, 2023-07-15:
The fix implemented in #475 :

https://github.com/winter-telescope/mirar/blob/5e68b3242117566ac84427787bc01b4179e80cc1/mirar/io.py#L240

does not work. It always returns true. As a consequence, the monitor is completely broken, because it tries to read in images which are partially written to disk. We need to reverse this, in a way that mef images pass.

PR #499 (2023-07-15): Fix bug in image complete check
Actually fix #469, undo #475
closed 2023-07-15 by robertdstein