winter-telescope/mirar #501: [BUG] WINTER photometric calibration is not reproducible

robertdstein, 2023-07-16:
**Describe the bug**
Repeating photometric calibration on a prototype test pipeline passes exactly sometimes, and on other occasions fails. The zeropoints are not always reproducible even on the same machine!

robertdstein, 2023-07-18:
This may be related to a seemingly random ordering that can occur in the pipeline. Maybe we should be sorting images at each step? We should for sure be sorting pre-stacking.

robertdstein, 2023-07-21:
Does not seem to have reoccurred. Will close, but reopen if needed.

robertdstein, 2023-07-26:
Seems to be happening again

robertdstein, 2023-07-26:
Hopefully fixed by #542

virajkaram, 2023-07-27:
This was related to the non-deterministic order with which Scamp ran. In the current version, the results are exactly reproducible on the same device. There are slight differences (at ~1 % level) between my computer and github-CI, even with the same versions of the softwares on both devices that I haven't been able to track down. 
These differences are within the tolerance levels we set for zeropoints, but we can track them down sometime later if we want to.

robertdstein, 2023-07-27:
Suggestion: print the CI values once, and then fix the tests to the Github CI values rather than your laptop. It'll give us more wriggle-room on tolerance.

robertdstein, 2023-07-27:
Anyway, this issue of internal non-reproducibility seems to indeed have been fixed by #542

closed 2023-07-27 by robertdstein