winter-telescope/mirar #514: [BUG]Split image names are no longer tagged by board_id

virajkaram, 2023-07-19:
**Describe the bug**
The split image nameing convention seems to have changed. They were previously being named `WINTERCamera_<utctime>_<board_id>_<subcoord_id>.fits`. 
Now looks like they are being named  `WINTERCamera_<utctime>_<board_id>_<extension_num>.fits`. This is an issue, because seems like the order in which extensions are written is not always the same. In reduced data from 20230713, the file names `WINTERcamera_20230711-083250-645_mef_5_0_1.fits_stack.fits` and `WINTERcamera_20230714-055212-760_mef_4_0_1.fits_stack.fits` both have `BOARD_ID=4` data, which is not good.

I liked the earlier approach because it was always deterministic (board_id was read from header), and suggest we revert back to that.

robertdstein, 2023-07-19:
I don't like the idea of hardcoding the mef loader to assume/check this winter-specific board ID is there. The current default seems reasonable, and should serve most use cases. Why not just do the `BASE_NAME_KEY` assignment in the load_winter_mef_image, after calling `open_mef_fits`?

virajkaram, 2023-07-19:
The way it was coded up previously - it defaulted to the extension number, and the user could  optionally provide a header key that uniquely identifies the extension (which doesn't seem too uncommon of a use-case - people using MEF images would likely have an identifier in the header).  
Also fine with the BASE_NAME_KEY if that is more elegant, regardless we should decide and fix this.

robertdstein, 2023-07-19:
Oh, well actually then let's do that original behaviour. We should just put exten_id as an argument for open_mef_image, default None. I can code this up this morning, it's an easy PR.

virajkaram, 2023-07-19:
Thanks, here's how it was originally if you want to just copy-paste - https://github.com/winter-telescope/mirar/blob/5ffbcce5a372e5537d9bd6ab2d87fe8aa7c4c518/mirar/processors/utils/image_loader.py#L213

PR #519 (2023-07-19): Assign board ID for mef with winter
Fix #514
closed 2023-07-19 by robertdstein