winter-telescope/mirar #611: [BUG] Mysterious monitor issue

robertdstein, 2023-08-11:
**Describe the bug**
```
Error for processor mirar.processors.database.database_inserter at 2023-08-10 17:00:16.521602 (local time): 
   File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 217, in apply_to_batch
    batch = self.apply(batch)
            ^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 243, in apply
    batch = self._apply(batch)
            ^^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 366, in _apply
    return self._apply_to_images(batch)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/database/database_inserter.py", line 54, in _apply_to_images
    assert len(res) == 1
           ^^^^^^^^^^^^^
AssertionError:  
  This error affected the following files: ['WINTERcamera_20230730-070048-558_mef.fits', 'WINTERcamera_20230730-070048-558_mef.fits'] 
This error was not a known error raised by mirar. 
```

virajkaram, 2023-08-11:
Possible clue: I've run into this when the code runs into a duplicate key conflict in the database, but there is no entry in the database that matches all the unique keys of the entry (e.g. for some reason if the `savepath` of a `stackid` has changed).

virajkaram, 2023-08-16:
Related to #618 ?

robertdstein, 2023-08-16:
Ah, yes, good point! Let's close for now, and reopen if it reoccurs.

closed 2023-08-16 by robertdstein