winter-telescope/mirar #613: [BUG] No way to see if WFCAM image downloaded from archive is deprecated

virajkaram, 2023-08-11:
**Describe the bug**
WFCAM has the concept of deprecated images that shouldn't be used. Problem is there is no way to query (afaik) whether an image is deprecated or not using astroquery or their [online form](http://wsa.roe.ac.uk:8080/wsa/getImage_form.jsp). The query returns deprecated and undeprecated images. The web-form displays the deprecated status so you can choose which to download, but unfortunately the astroquery does not return this info. I was relying on cuts on zeropoints and seeing to exclude the deprecated images from stacking, but obviously this does not take care of all cases.

Through magical means, I have acquired the deprecated status of all WFCAM images, and ingested only the undeprecated image files on the winter machine, so we don't have to worry about bad images creeping in for winter ops. We should add a xmatch to this file to remove deprecated images from online queries as well.

robertdstein, 2023-08-11:
Maybe this is worth an issue on astroquery?

virajkaram, 2023-08-12:
Addressed by #614

virajkaram, 2023-08-14:
More permanent solution here - https://github.com/astropy/astroquery/pull/2809