winter-telescope/mirar #643: [BUG] swarp quality is questionable for SEDMv2

saarahhall, 2023-08-24:
**Describe the bug**
The current quality of Swarp stacks (at least for SEDMv2) is questionable. I say this because:
1. stacked SEDMv2 images do not "see" as deep as we expect them to
2. just by looking at and blinking between Swarp's intermediate resampled frames, alignment has room for improvement

I believe the issue is that Swarp isn't reading astrometry.net's WCS solutions (the SIP terms specifically). This has come up before, and it's why we stopped using Scamp.

**Plan to fix the bug**
I will attempt to follow the solution described by J. Findlay [here](http://physics.uwyo.edu/~WIRO/DoublePrime/WDPreductions.pdf). AKA I will run a-net without SIP solutions, then sextractor, then scamp, then swarp. I don't recall if this has been tried before since adding the a-net processor.

**Additional context**
I realize this isn't a bug with any particular processor, I'm just announcing it here in case anyone else has noticed poor stacks.

virajkaram, 2023-08-24:
@saarahhall this is what we do for WINTER. I think none of the astromatic packages are capable of interpreting `SIP` distortion coefficients (that come out of astrometry.net). 

Hopefully the solution you proposed works well for you, but an alternate solution to this problem is [this](https://pypi.org/project/sip_tpv/ ) package that converts the SIP coefficients from astrometry.net to TPV coefficients that swarp, etc. can understand. 

We should have this documented and also prohibit running Swarp with just AstrometryNet in mirar.

virajkaram, 2023-08-24:
Here's what we do for WINTER - https://github.com/winter-telescope/mirar/blob/4df57f8f1b473034648065db044cd24bfb9b534f/mirar/pipelines/winter/blocks.py#L298

saarahhall, 2023-08-24:
thanks for this insight! glad there are multiple ways to approach this, I’ll try to mirror WINTER first

saarahhall, 2023-08-28:
`(base) dhcp-10-105-14-182:notebooks saarahhall$ solve-field /Users/saarahhall/Desktop/research/outputdata/sedmv2/20230630/a-net/temp_kped_20230630_093036_ZTF_J2130+4420_sg_o_0000_0.fits --dir /Users/saarahhall/Desktop/research/outputdata/sedmv2/20230630/a-net --new-fits /Users/saarahhall/Desktop/research/outputdata/sedmv2/20230630/a-net/kped_20230630_093036_ZTF_J2130+4420_sg_o_0000_0.fits --overwrite --out kped_20230630_093036_ZTF_J2130+4420_sg_o_0000_0 --no-tweak  --scale-high 0.11666667  --scale-low 0.08333333 --scale-units degw --downsample 2 -X X_IMAGE -Y Y_IMAGE -s MAG_AUTO --sort-ascending --ra 322.736 --dec 44.3462 --radius 5.0 
Reading input file 1 of 1: "/Users/saarahhall/Desktop/research/outputdata/sedmv2/20230630/a-net/temp_kped_20230630_093036_ZTF_J2130+4420_sg_o_0000_0.fits"...
Found an existing WCS header, will try to verify it.
Extracting sources...
Downsampling by 2...
simplexy: found 448 sources.
Traceback (most recent call last):
  File "/opt/homebrew/bin/removelines", line 15, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/homebrew/Cellar/astrometry-net/0.93/libexec/lib/python3.11/site-packages/astrometry/util/removelines.py", line 77, in main
    return removelines(infile, outfile, xcol=opt.xcol, ycol=opt.ycol,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/astrometry-net/0.93/libexec/lib/python3.11/site-packages/astrometry/util/removelines.py", line 46, in removelines
    ix = hist_remove_lines(T.get(xcol), 1, 0.5, logcut=-cut)
                           ^^^^^^^^^^^
  File "/opt/homebrew/Cellar/astrometry-net/0.93/libexec/lib/python3.11/site-packages/astrometry/util/fits.py", line 316, in get
    return self.getcolumn(name)
           ^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/astrometry-net/0.93/libexec/lib/python3.11/site-packages/astrometry/util/fits.py", line 306, in getcolumn
    return self.__dict__[name]
           ~~~~~~~~~~~~~^^^^^^
KeyError: 'X_IMAGE'
augment-xylist.c:591:backtick Failed to run command: /opt/homebrew/bin/removelines -X X_IMAGE -Y Y_IMAGE /tmp/tmp.xyls.P2uYsX /tmp/tmp.removelines.0rNy55
 ioutils.c:568:run_command_get_outputs Command failed: return value 1
`