winter-telescope/mirar #646: [BUG] Mysterious scamp error

virajkaram, 2023-08-26:
**Describe the bug**
```
Error for processor mirar.processors.astromatic.scamp.scamp at 2023-08-25 18:33:47.738924 (local time):
   File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 217, in apply_to_batch
    batch = self.apply(batch)
            ^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 243, in apply
    batch = self._apply(batch)
            ^^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/base_processor.py", line 366, in _apply
    return self._apply_to_images(batch)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/loki/code/mirar/mirar/processors/astromatic/scamp/scamp.py", line 171, in _apply_to_images
    run_scamp(
  File "/data/loki/code/mirar/mirar/processors/astromatic/scamp/scamp.py", line 64, in run_scamp
    execute(scamp_cmd, output_dir=output_dir, timeout=np.max([60.0, timeout_seconds]))
  File "/data/loki/code/mirar/mirar/utils/execute_cmd.py", line 311, in execute
    run_local(cmd, output_dir=output_dir, timeout=timeout)
  File "/data/loki/code/mirar/mirar/utils/execute_cmd.py", line 117, in run_local
    raise ExecutionError(msg) from err
ExecutionError: Execution Error found when running with command:

 'scamp @/data/loki/data/winter/20230726/scamp/WINTERcamera_20230727-060411-957_mef_0_0_0.fits_scamp_list.txt -c /data/loki/code/mirar/mirar/pipelines/winter/config/files/astrom.scamp -ASTREFCAT_NAME /data/loki/data/winter/20230726/scamp/WINTERcamera_20230727-060411-957_mef_0_0_0.local.cat -VERBOSE_TYPE QUIET -SOLVE_PHOTOM N'

This yielded a return code of -11. The following traceback was found:

> WARNING: FGROUP_RADIUS keyword unknown

> WARNING: REF_PORT keyword unknown

> WARNING: CDSCLIENT_EXEC keyword unknown

> WARNING: Obsolete configuration; replacing cocat1.u-strasbg.fr with vizier.u-strasbg.fr

sh: convert: command not found

> WARNING: phot_rp_mean_mag_error parameter not found in catalog WINTERcamera_20230727-060411-957_mef_0_0_0.local.cat

  This error affected the following files: ['WINTERcamera_20230727-060411-957_mef.fits', 'WINTERcamera_20230727-060617-917_mef.fits', 'WINTERcamera_20230727-060823-736_mef.fits', 'WINTERcamera_20230727-061030-050_mef.fits', 'WINTERcamera_20230727-061236-361_mef.fits']
This error was not a known error raised by mirar.
```

**Additional context**
Probably a result of 
```
----- Reference catalogs:

> Loading Catalog WINTERcamera_20230727-062752-730_mef_2_0_1.local.cat...
> WARNING: phot_rp_mean_mag_error parameter not found in catalog WINTERcamera_20230727-062752-730_mef_2_0_1.local.cat

 0 astrometric references loaded from WINTERcamera_20230727-062752-730_mef_2_0_1.local.cat
 Group  1:        0 standard found in file (1 band)

  
----- Astrometric clipping:

> Making cross-identifications in group 1
Segmentation fault (core dumped)

```

virajkaram, 2023-08-27:
This is caused by the catalog not overlapping with the images. Have implemented a fix, but this has revealed another problem - currently all images of a field are grouped and stacked together. 
However, on some nights, we get two observations (each with 5 dithers) separated by several hours on the same night, that are currently being stacked into one. Sometimes the two sets of observations are offset by a few degrees for some reason, which causes scamp to complain in a similar fashion as above. 
Additionally, in general I think we don't want to stack multiple observations like these together, so we need to come up with a way of distinguishing them.

robertdstein, 2023-08-27:
Duplicate of https://github.com/winter-telescope/mirar/issues/596

virajkaram, 2023-08-28:
This is the source of random ExecutionErrors. Solution : a combination of OBSHISTID and QCOMMENT in the headers.

closed 2024-01-04 by virajkaram