winter-telescope/mirar #698: [BUG] WINTER pipeline creates q3c extensions every time it is run

virajkaram, 2023-11-10:
**Describe the bug**
Every time WINTER pipeline is run, it creates q3c extensions on a table. This takes forever to run (also possibly scales with the number of entries present in the db), causing the pipeline to stall for many many minutes before it actually starts reducing the images. Also this is is not necessary and we shouldn't be doing it. We can probably check whether the extension exists and make it only if it does not.

**Additional context**
Add any other context about the problem here.

robertdstein, 2023-11-27:
I agree with your suggested solution!

virajkaram, 2024-01-05:
Solve

virajkaram, 2024-01-05:
FYI this had created several (hundred) copies of the same index on the tables in the database. big mess.