winter-telescope/mirar #751: [FEATURE] WINTER candidates q3c queries take forever

virajkaram, 2024-01-31:
**Is your feature request related to a problem? Please describe.**
The most time consuming step (~1 hour) for winter pipeline currently is q3c queries to the database for every single candidate.  It doesn;t help that the databases don't run parallely.

**Describe the solution you'd like**
No obvious single solution, will require experimenting to see what (if anything) is better. Possible solution is doing a single large database query per image and then crossmatching using astropy.

robertdstein, 2024-01-31:
Just a comment that we initially restricted multiple database connections because we were hitting "too many connections" errors. So bear that in mind.

robertdstein, 2024-02-21:
Partially fixed, or at least outdated after #771 and #763. We no longer do q3c queries on candidates at all. Let's revisit if it becomes an issue for the source table.

closed 2024-02-21 by robertdstein