winter-telescope/mirar #817: [BUG] Pre-commit hooks require python3.10 but installation docs say to install python3.11

jamiesoon, 2024-03-27:
**Describe the bug**
Pre-commit hooks require python3.10 but installation docs say to install python3.11

**Expected behaviour**
Either pre-commit hooks updated to 3.11 or docs instructions should be updated to install 3.10

robertdstein, 2024-03-28:
I think the pre-commit hooks do NOT require any version of python? At least, I have been using this with python3.11 without issue. 

I do see this line:
https://github.com/winter-telescope/mirar/blob/bcd254c82b1ae063895215b75fc642b04d56049b/.pre-commit-config.yaml#L2
but I believe that is a default. I don't think this line implies pre-commit requires python3.10. Did you encounter a specific bug?

(And feel free to PR this to 3.11)

jamiesoon, 2024-04-05:
The pre-commit does require the version of python listed in .pre-commit-config.yaml. It looks like it creates a virtual environment based on that version. See error below.

`An unexpected error has occurred: CalledProcessError: command: ('{conda_path}/conda/mirar/bin/python', '-mvirtualenv', '/home/jamies/.cache/pre-commit/repo556gsn8c/py_env-python3.10', '-p', 'python3.10')`

When you initially did your pre-commit installation it may have been with python3.10 which is why it's still working. If you run pre-commit install again with a python3.11 conda environment it will probably break.

I will create a PR to bump it to python3.11

PR #828 (2024-04-07): Bump pre-commit to python3.11
Fixes #817
closed 2024-04-07 by jamiesoon