winter-telescope/mirar #881: [FEATURE] Fail images with no stars

virajkaram, 2024-05-08:
Suggestion from Mansi - Use the Sextractor catalogs to identify images with no stars in them and throw them away. Will be useful to diagnose failures due to clouds/mirror-cover/dome issues. Will take some experimentation to figure out what flags and parameters to use to select real stars from the Sextractor catalog without being too strict.

robertdstein, 2024-05-08:
This is definitely not straightforward. I am pretty sure we already have a check for "not enough stars", or at least did for autoastrometry. These darks currently have ~500 stars recovered though, presumably due to non-linearity, so the cuts would need to be extremely aggressive.

virajkaram, 2024-05-08:
We do not run autoastrometry for WINTER. For astrometry.net, the sextractor thresholds are deliberately set to low values, and have very minimal quality cuts, because anet  does a good job of figuring out which stars are real and use them for solving astrometry, and can use low S/N stars to build some confidence in the solution. The detected "stars" in the darks are due to hot-pixels and masking-edge effects, which could be filtered out using FWHM, SNR and FLAGS parameters in the catalogs. 

However, I am apprehensive about how useful this will be.  I would mark this as wishlist, https://github.com/winter-telescope/mirar/issues/878 will definitely give the most effective filtering check and should take priority.